seemoo-lab/bcm-rpi3 #5: Won't compile the wi-fi modded drivers: please add them to the repository.

matt6hub, 2016-09-22:
Not to be rude, but my make always stops compiling with too many errors. So, could you please upload the modded wi-fi drivers for the Raspberry Pi 3? Thank you in advance.

DanielAW, 2016-09-22:
Hi,
on which system did you try to compile our code? Please keep in mind it is supposed to be cross-compiled, e.g. on a Linux host. 
If we don't know whats broken we can't fix it :wink:. Can you copy at lease the first few errors you get while compiling?

matt6hub, 2016-09-22:
On an X86 PC running ubuntu MATE 16.04
root@ubuntu:/home/pinet/bcm-rpi3/firmware_patching# cd monitor_mode
root@ubuntu:/home/pinet/bcm-rpi3/firmware_patching/monitor_mode# make
make -C ../../kernel/ bcm2709_defconfig
make[1]: Entering directory '/home/pinet/bcm-rpi3/kernel'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
make[1]: Leaving directory '/home/pinet/bcm-rpi3/kernel'
make -C ../../kernel/
make[1]: Entering directory '/home/ubuntu/bcm-rpi3/kernel'
/home/ubuntu/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /home/ubuntu/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
scripts/kconfig/conf  --silentoldconfig Kconfig
/home/ubuntu/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /home/ubuntu/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/msi.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  WRAP    arch/arm/include/generated/asm/seccomp.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/sembuf.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/shmbuf.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/termios.h
  WRAP    arch/arm/include/generated/asm/timex.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
/home/ubuntu/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /home/ubuntu/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
Kbuild:44: recipe for target 'kernel/bounds.s' failed
make[2]: *** [kernel/bounds.s] Error 2
Makefile:987: recipe for target 'prepare0' failed
make[1]: *** [prepare0] Error 2
make[1]: Leaving directory '/home/ubuntu/bcm-rpi3/kernel'
Makefile:8: recipe for target 'brcmfmac/brcmfmac.ko' failed
make: *** [brcmfmac/brcmfmac.ko] Error 2

DanielAW, 2016-09-23:
It seems like the kernel is not build in our case and building it from the "wrong" directory causes the error. Are you on the latest commit on the master branch? We fixed this problem in one of the previous commits.
Alternatively is it possible to build the kernel in the kernel directory first:
```
cd kernel/
make
cd ../firmware_patching/monitor_mode/
make
```

matt6hub, 2016-09-25:
About the wi-fi drivers, I think it would be better for someone with the time and patience to implement the wi-fi driver monitor mode patch for end users please. Thank you in advance!

DanielAW, 2016-09-25:
Honestly, I'm not sure what you mean. Is the original issue solved from you side? If this is the case, I would like to close this issue.

mastahofdesastah, 2016-09-26:
Hi

Itried to compile on multiple devices/OS and get this error:

make -C ../../kernel/ bcm2709_defconfig
make[1]: Entering directory '/root/bcm-rpi3/kernel'
#
# configuration written to .config
#
make[1]: Leaving directory '/root/bcm-rpi3/kernel'
make -C ../../kernel/
make[1]: Entering directory '/root/bcm-rpi3/kernel'
/root/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /root/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
scripts/kconfig/conf  --silentoldconfig Kconfig
/root/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /root/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[2]: 'include/generated/mach-types.h' is up to date.
  CC      kernel/bounds.s
/root/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /root/bcm-rpi3/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
Kbuild:44: recipe for target 'kernel/bounds.s' failed
make[2]: *** [kernel/bounds.s] Error 2
Makefile:987: recipe for target 'prepare0' failed
make[1]: *** [prepare0] Error 2
make[1]: Leaving directory '/root/bcm-rpi3/kernel'
Makefile:8: recipe for target 'brcmfmac/brcmfmac.ko' failed
make: *** [brcmfmac/brcmfmac.ko] Error 2

This is while compiling in Kali

E3V3A, 2016-09-26:
@DanielAW Is there any good reason (apart academic) why you do not provide the already compiled `brcmfmac.ko` module? It seem quite painful to have to compile the entire kernel just to get this file. (You do include the *original* one.) Also, it is not clear from your README what other dependencies there are for the Kernel (if any) to support your patch. You say: 

> Switching channels ... it works now, but it needs a patch in the kernel, see ...
> Injection does not work (due to problems with the brcmfmac driver)

Please clarify. (Also, what exactly are the problems for injection?)

DanielAW, 2016-09-26:
> @DanielAW Is there any good reason (apart academic) why you do not provide the already compiled brcmfmac.ko module?

You mean a patched one for monitor mode? The DD image contains a compiled kernel module. The problem with prebuild kernel modules is, that the have to match the [kernel version magic](http://www.tldp.org/LDP/lkmpg/2.6/html/x380.html) of the kernel your are currently using. You will be unable to load a prebuild kernel module if your kernel version is slightly diffirent. 

Switching channels is more difficult than it looks in the patch you mention. 8f4b7501dedf72306c471e4962478b8ffe91d7a8 is only the _kernel_ patch. We had to add a function which allows the switching of channels in the _driver_ too, see https://github.com/seemoo-lab/bcm-rpi3/commit/9b02bd6d493a2944565098111d29d47b0225300d .

I opened a seperate issue for the injection support, see #6

@mastahofdesastah you can try the solution mentioned here http://unix.stackexchange.com/a/45784 for the kconfig script, but in general I can't recommend trying to build anything kernel related on kali. I've experienced some wired problems with the past :disappointed:

E3V3A, 2016-09-26:
> The problem with prebuild kernel modules is, that the have to match the kernel version magic of the kernel your are currently using. You will be unable to load a prebuild kernel module if your kernel version is slightly diffirent. 

Yeah, so I was (hoping) thinking that one could try to keep updated kernel modules for the official Jesse releases, which I think are not changing that often. (Or are they?) In that case I'd imagine we only need to build a few (RPi-0/1/2/3 A/B), then if the user fails to run it, he can just run `apt dist-upgrade` to get a match.

E3V3A, 2016-09-26:
BTW. Why is it not recommended or possible to build directly on the (let's say) RPi3? Size or memory issues?

XePeleato, 2016-09-26:
@E3V3A It's possible, maybe by checking the architecture, and if it detects that it's being buit on the RPi, it just avoids all of the cross compiling.

mastahofdesastah, 2016-09-28:
I tried to compile on a Raspberry with Kali, an PC 32/64 with windows, and on a 32bit debian machine.
Now after you said it doesent work on kali, i tried to compile on Ubuntu. I get this error:
`make -C ../../kernel/ bcm2709_defconfig
make[1]: Entering directory '/home/osboxes/Desktop/bcm-rpi3-master/kernel'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
make[1]: Leaving directory '/home/osboxes/Desktop/bcm-rpi3-master/kernel'
make -C ../../kernel/
make[1]: Entering directory '/home/osboxes/Desktop/bcm-rpi3-master/kernel'
/home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
scripts/kconfig/conf  --silentoldconfig Kconfig
/home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/msi.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  WRAP    arch/arm/include/generated/asm/seccomp.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/sembuf.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/shmbuf.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/termios.h
  WRAP    arch/arm/include/generated/asm/timex.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
/home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
Kbuild:44: recipe for target 'kernel/bounds.s' failed
make[2]: *** [kernel/bounds.s] Error 2
Makefile:987: recipe for target 'prepare0' failed
make[1]: *** [prepare0] Error 2
make[1]: Leaving directory '/home/osboxes/Desktop/bcm-rpi3-master/kernel'
Makefile:8: recipe for target 'brcmfmac/brcmfmac.ko' failed
make: *** [brcmfmac/brcmfmac.ko] Error `

and when i try to compile directly the kernel i get this:
`/home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[1]: 'include/generated/mach-types.h' is up to date.
  CC      kernel/bounds.s
/home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: /home/osboxes/Desktop/bcm-rpi3-master/buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
Kbuild:44: recipe for target 'kernel/bounds.s' failed
make[1]: *** [kernel/bounds.s] Error 2
Makefile:987: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
`

Can you say on which distro/kernel it will build?

mastahofdesastah, 2016-09-28:
And the solution 
http://unix.stackexchange.com/a/4578
Describes to put the shebang on the first line. On the file setup_env.sh it is on the first line. So nothing to change. Or what you meant with this hint?

XePeleato, 2016-09-28:
@mastahofdesastah Because you can't crosscompile it, and there's no need, so remove the toolchain and build the kernel without cross compilation and it should work.

XePeleato, 2016-09-28:
@mastahofdesastah The point of cross compiling is to compile the C code into Armv7 instructions. The instructions read by a x64 Desktop computer are different than the ones read by the RPi. So yeah, it tells you to cross compile because it's meant to be built on a computer. I think that the cross compiler is specified in 'setup_env'.

mastahofdesastah, 2016-09-28:
I found it. Its in the Makefile. Removed and its compiling right now since over 10min

E3V3A, 2016-09-28:
@mastahofdesastah Let us know how it goes, and what you did to get it to work. Will be very helpful for people who don't have another VM or Linux box to compile on. I guess the main reason for wanting to do cross compilation is to reduce compile time, which means not more than a few minutes on a half-descent PC.

mastahofdesastah, 2016-09-28:
still compiling on rpi3. 2h+

stdclass, 2016-09-28:
Any updates so far? Getting the same error message on Raspbian / RPi3

mastahofdesastah, 2016-09-28:
Sorry wasnt home.
Compiled with errors an took over 4h
in the folder /brcmfmac now i have the brcmfmac.ko, but i cant find the "brcmfmac43430-sdio.bin". Maybe because of the compiling error. I will get a look later in the evening...
Here the compiling errors:
`make[1]: Leaving directory '/broadcom/bcm-rpi3/kernel'
make -C ../../kernel/ M=$PWD/brcmfmac -j2
make[1]: Entering directory '/broadcom/bcm-rpi3/kernel'
  LD      /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/built-in.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/cfg80211.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/chip.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/fwil.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/fweh.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/fwsignal.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/p2p.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/proto.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/common.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/core.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/firmware.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/feature.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/btcoex.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/vendor.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/bcdc.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/sdio.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/bcmsdh.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/usb.o
  CC [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/of.o
  LD [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/brcmfmac.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/brcmfmac.mod.o
  LD [M]  /broadcom/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/brcmfmac.ko
make[1]: Leaving directory '/broadcom/bcm-rpi3/kernel'
../../buildtools/arm-eabi-4.7/bin/arm-eabi-gcc -Wall -Werror -O2 -nostdlib -nostartfiles -ffreestanding -mthumb -march=armv7-m -ffunction-sections -fdata-sections -c patch.c -o patch.o
../../buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: 1: ../../buildtools/arm-eabi-4.7/bin/arm-eabi-gcc: Syntax error: ")" unexpected
Makefile:16: recipe for target 'patch.o' failed
make: *** [patch.o] Error 2`

XePeleato, 2016-09-28:
@mastahofdesastah That error mentions the cross compiler, so it's being called somewhere. Anyway, I am not sure but in my opinion there's something weird about the toolchain included here.

I'll try to reproduce your issue and submit a PR if I find a solution.

mastahofdesastah, 2016-09-28:
@XePeleato thanks for the help
Yea i assume the same, but cant find it where the second call is located.
Sorry im such a noob, what you mean with a PF?

matthiasseemoo, 2016-09-29:
To compile the WiFi firmware for the Cortex-M3 processor, we use the ARM Embedded Toolchain (https://developer.arm.com/open-source/gnu-toolchain/gnu-rm) that is only available for x86 Linux. You can try to compile the firmware patch with the same compiler you used to compile the kernel, if it supports the armv7-m architecture. To do this, just adjust the CC definition in the Makefile of the monitor_mode patch. Keep in mind, that in the future your gcc compile should have plugin support enabled as we extended the compiler with an "at" attribute to simplify writing firmware patches. The plugin is already available in the bcm-public repository for the Nexus 5: https://github.com/seemoo-lab/bcm-public

mastahofdesastah, 2016-09-29:
@matthiasseemoo Thanks for clarification. Sorry im new in the linux/programming world. Can you pint out what to change in the Makefile?
I believe you refeer to this line:
CC=../../buildtools/arm-eabi-4.7/bin/arm-eabi-

E3V3A, 2016-09-29:
(**A**) In the `./firmware_patching/monitor_mode/Makefile` the [first line](https://github.com/seemoo-lab/bcm-rpi3/blob/master/firmware_patching/monitor_mode/Makefile#L1) says: 
```
CC=../../buildtools/arm-eabi-4.7/bin/arm-eabi-
```

(**B**) In the `/setup_env.sh`, [line 6-7](https://github.com/seemoo-lab/bcm-rpi3/blob/master/setup_env.sh#L6-L7) says:
```
 export CC=$(pwd)/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-
 export CROSS_COMPILE=$(pwd)/buildtools/arm-eabi-4.7/bin/arm-eabi-
```

So I believe you need to replace the `CC` line in part (A) with that in part (B), and remove or comment out the `CROSS_COMPILE` from (B). I didn't try this though...

E3V3A, 2016-09-29:
Perhaps @plus44 who did it in #2 could help telling us what he did.

mastahofdesastah, 2016-09-30:
thanks for al the help. I modified all akefiles to

DanielAW, 2016-10-27:
We now support building the modified driver and fw patches on the RPI3 itself. I will close this issue for now.

closed 2016-10-27 by DanielAW