seemoo-lab/bcm-rpi3 #6: Injection support

DanielAW, 2016-09-26:
The current monitor mode implementation does not allow transmitting any frames. The problem is that frames never arrive at the firmware. It seems that the frames don't even arrive at the driver.
I would assume that frames need to pass the **brcmf_netdev_start_xmit()** function in _core.c_. According to my tests this function is never called if I send a frame to the socket of the Wifi interface. There are also no error messages on the involved system calls.

This needs further investigation.

E3V3A, 2016-09-28:
> frames never arrive at the firmware

This doesn't sound right, frames could impossibly avoid the firmware. 

> It seems that the frames don't even arrive at the driver.

Hmm, unless the driver is built-in elsewhere or the device is using a different driver or interface all-together.

@DanielAW  What *tests* have you done? Can you point to more details or code so others with more experience may chime in?

E3V3A, 2016-09-28:
Not sure if this helps, but looking in the file: `/lib/firmware/brcm/brcmfmac43430-sdio.txt` we have the lines:
```bash
...
# muxenab: 0x1 for UART enable, 0x2 for GPIOs, 0x8 for JTAG
muxenab=0x1
...
```
This seem to indicate that the WiFi is using the UART for communication.

DanielAW, 2016-09-29:
> This doesn't sound right, frames could impossibly avoid the firmware.
I'm 100% sure the frames do not arrive the firmware. We identified the function responsible for handling incomming packets from the SDIO bus. These SDIO packes are also marked as data or management packets. Not a single data frame is arriving at the firmware in the current monitor mode implmenetation.

> @DanielAW What tests have you done? 
I simply send crafted frames to a raw socket using libpcap. This method works on the Nexus 5. 

This leaves me with the conclusion that it has something to do with the driver or any interface in the kernel.
My guess would be the cfg80211 interface (interface between kernel and driver). The interface maybe needs to somehow know that we are associated with a station, frames are only forwarded to the driver if this is the case.

Regarding a debugging interface:
The debug pins of the bcm4339 in the Nexus 5 are most likely not connected to the PCB. I would assume this is also the case in the Raspberry Pi 3.

E3V3A, 2016-09-29:
@DanielAW  
I've been looking at some documents for similar devices as can be briefly compared [here](https://wikidevi.com/wiki/Broadcom#SDIO/Other). Then I searched for BCM43362 at the *Cypress* site, which gave me [this PDF](http://www.cypress.com/file/298336/download). There it reads: 

> In an integrated single-chip combo solution, the BCM43XX implements 
highly sophisticated enhanced collaborative coexistence hardware 
mechanisms and algorithms to enable the WLAN and Bluetooth to operate 
simultaneously and to ensure maximum medium access time, high 
throughput, and audio quality. Collaborative coexistence between WLAN 
and Bluetooth is implemented according to IEEE 802.15.2 Packet Traffic 
Arbitration (PTA) and through Broadcom's Enhanced Coexistence Interface 
(ECI). ECI augments PTA signaling by enabling exchange of additional 
information required for implementing more advanced collaborative 
coexistence methods. As a result, enhanced overall quality for 
simultaneous voice, video, and data transmission on an embedded system 
is achieved. 
> 
> In a discrete, separated WLAN and Bluetooth solution, or other external 
radios such as LTE and Zigbee, collaborative coexistence supports 
through SECI, UART, or 3-wire hardware signaling through a unique set of 
Serial Enhanced Coexistence Interface (SECI) or Global Coexistence 
Interface (GCI) protocols. Using the prioritization approach between 
data types and applications, optimum performance can be achieved, 
resulting in maximum WLAN throughput, voice quality, and link 
performance.

So yeah, this is definitely a little more complicated than first thought. But I guess the point of SECI is to make it simpler! Looking at the block diagrams of the BCM43438 is interesting... 

[image]

E3V3A, 2016-10-04:
@DanielAW Any news or ideas?

DanielAW, 2016-10-05:
I can add a few things to the list which did not work:
* Starting the TX queue right after the interface is set up:
`netif_start_queue(ndev)`
* set the interface to connected: `set_bit(BRCMF_VIF_STATUS_CONNECTED, &ifp->vif->sme_state);`

I also looked into how a connection to an access point is established, I think it works like this:

1. The function brcmf_cfg80211_connect() gets called via the `.connect` function pointer in the  `wl_cfg80211_ops` struct if a new connection should be established
2. It sends the IOCTL 26 (BRCMF_C_SET_SSID) to the firmware
3. The firmware in return sends the JOIN event (via the `wlc_join()` function) to the driver which is handled by the `brcmf_notify_connect_status()` function.
4. This function calles `brcmf_bss_connect_done()` which presumably does some stuff to the interface which allows packets to arrive the driver [brcmf_netdev_start_xmit()]

We now want to tell the driver it is connected by a non existing station by hooking the function responsible for handling IOCTL 26 in the firmware, and sending the JOIN event to the driver if the firmware receives a specific SSID.

I would welcome an easier solution, but as long as I do not fully understand why the interface does silently drop my packes if I'm not connected, this is the only thing I currently think which could work  :confused:

E3V3A, 2016-10-05:
> I do not fully understand why the interface does silently drop my packes if I'm not connected,

Can it be a design feature in order to conserve power? After all its a chipset mostly meant for embedded devices. A low level patch work-around would be preferred but on a simpler upper level, perhaps it could be enough to connect to a virtual dummy interface? Now, I don't know if that is even possible, just throwing some ideas around.

DanielAW, 2016-10-05:
Short node: I was able to track the problem down to this function call https://github.com/seemoo-lab/bcm-rpi3/blob/master/kernel/net/core/dev.c#L2915. It returns 2 if I'm not connected to a access point, which in return causes the frame to be dropped.

DanielAW, 2016-10-05:
Ok, I got it. This line needs to go: https://github.com/seemoo-lab/bcm-rpi3/blob/master/firmware_patching/monitor_mode/brcmfmac/core.c#L677

Now we can continue to implement the firmware side, stay tuned :-)

E3V3A, 2016-10-05:
Yeah [this](https://github.com/seemoo-lab/bcm-rpi3/blob/master/kernel/net/sched/sch_sfq.c#L68-L81) queuing stuff is funky. Not sure what they mean with *flow* here, but perhaps packets are not sent/processed until you have 127? 

```c
/*
	IMPLEMENTATION:
	This implementation limits :
	- maximal queue length per flow to 127 packets.
	- max mtu to 2^18-1;
	- max 65408 flows,
	- number of hash buckets to 65536.
	It is easy to increase these values, but not in flight.  */

#define SFQ_MAX_DEPTH		127 /* max number of packets per flow */
#define SFQ_DEFAULT_FLOWS	128
#define SFQ_MAX_FLOWS		(0x10000 - SFQ_MAX_DEPTH - 1) /* max number of flows */
#define SFQ_EMPTY_SLOT		0xffff
#define SFQ_DEFAULT_HASH_DIVISOR 1024
```

So do we have a party soon?

E3V3A, 2016-10-07:
@DanielAW Awesome patch: 
https://github.com/seemoo-lab/bcm-rpi3/commit/f52aa28efca086c6749fe9fe875b637eac8a7793 

How can we best test it? Then we also need to update the README.

DanielAW, 2016-10-07:
Hi @E3V3A ,
currently there is the problem that the chip crashes if too many packets arrive. Thats why I do not want to announce it jet. If this problem is solved I update the README and create a new dd image. 

Sending packets with time in between is no problem. But `aireplay -9 wlan0` for example sends lots of packets which causes the chip to crash. This is also the case if I inject the packets manually (with my own C program).

E3V3A, 2016-10-07:
@DanielAW Perhaps it is a scheduling (as in previously mentioned snippet) or size problem? Whatever it is, something (buffer?) is getting overloaded and can't catchup, perhaps triggering a watchdog/reset? When you say *crashing*, how does that show? Is there a way you can count the packets and measure the time just before crash? I.e. set you test program to send X number of packets within Y seconds and slowly reduce Y until crash.

E3V3A, 2016-10-07:
**Maybe** because of the SECI/GCI, an overload of packets get dropped to make room for BT, or in our (patched) case not, lead to a reset?

DanielAW, 2016-10-07:
> When you say crashing, how does that show?
The chip does not respont anymore, neither sending nor receiving works. 

The internal console shows the following (you can get this information via `cat /proc/nexmon_consoledump`): 

```
TRAP c(6fdf8): pc dafa, lr dc51, sp 6fe4c, cpsr 2000000c, spsr 21000010
000819.311   r0 b88, r1 b88, r2 6fee8, r3 0, r4 6ad6c, r5 ffff, r6 4e76c
```
I **think** 0xc is a prefetch abort. 0xdafa is in the  wlc_send_q() which in return gets called by the wlc_sendctl() function.

E3V3A, 2016-10-07:
Unfortunately that low level errors are rather foreign and a bit over my head.  Sorry i can't be of more help here.

E3V3A, 2016-10-10:
Hi Daniel, any more indights. The 1Mbit [rate limit](https://github.com/seemoo-lab/bcm-rpi3/blob/5e26db21934d6a02236ffe797678f6931cc19fbe/firmware_patching/monitor_mode/patch.c#L180), is that something you determined from experiment or part of original firmware. 1Mb seem kind of low...

commit 30560e56 (2016-10-11): removed arm-eabi-4.7 compiler, it is superseded by the gcc-arm-none-eabi-5_4-2016q2 of the new patching framework; fixes #6 injection support works now without using the tx queue
closed 2016-10-11 by DanielAW