seemoo-lab/bcm-rpi3 #7: Update erroneous README

E3V3A, 2016-09-28:
I think your README file has an error in it, you write: 
> repository contains firmware patches for the **BCM43438** WiFi chip used in the Raspberry Pi 3

AFAICT, the WiFi chip should be **BCM43430**. Would be interesting to know the difference between these two chips though.

DanielAW, 2016-10-05:
What is your source? The string `BCM43430` contained in the firmware is not enough evidence IMHO. The guys in this ticket  seem to have contact to broadcom and they also talk about the BCM4343**8** https://github.com/raspberrypi/linux/issues/1325.

E3V3A, 2016-10-05:
Actually I'm starting to doubt that name too. I think it could have been either an unintentional typo or deliberate misnomer, propagated into Linux device driver source code.

DanielAW, 2016-10-21:
No new information so far, closed for now

closed 2016-10-21 by DanielAW

E3V3A, 2016-10-23:
Doing strings on the file `BCM43430A1.hcd` gives this:
```bash
# strings -a /lib/firmware/BCM43430A1.hcd |grep BCM
(BCM43438A1 37.4MHz Raspberry Pi 3-0043O
```
This is different from what's written in the file `brcmfmac43430-sdio.bin`:
```bash
# strings -a brcmfmac43430-sdio.bin |grep roml
43430a1-roml/sdio-g-p2p-pool-pno-pktfilter-keepalive-aoe-mchan-tdls-proptxstatus-ampduhostreorder-lpc-sr-bcmcps 
Version: 7.45.41.26 
CRC: 5932ca06 
Date: Fri 2016-05-27 00:15:32 
PDT Ucode Ver: 1043.2060 
FWID: 01-df77e4a7
```

DanielAW, 2016-10-23:
This makes sense, thanks :thumbsup:

E3V3A, 2016-11-17:
@DanielAW  Cypress just acquired Broadcom, and released all their datasheets. The revised edition for the BCM 43438 is [here](http://www.cypress.com/file/298076/download).

DanielAW, 2016-11-17:
I think the "only" bought the IOT from Broadcom. This happend a few month ago. But it is still nice to have the correct datasheet now, thanks :wink: