seemoo-lab/bcm-rpi3 #8: Compilation error: implicit declaration of wlc_bsscfg_find_by_wlcif and wlc_sendctl

aagallag, 2016-10-07:
When running make on master, I encounter a build error.  I noticed that if I revert the commit: f52aa28efca086c6749fe9fe875b637eac8a7793, the build process completes successfully.  Is this behavior currently expected or is there something wrong with my environment?

Thanks!
-Aaron

```
make[1]: Leaving directory '/home/aaron/Development/Github/bcm-rpi3/kernel'
make -C ../../kernel/ M=$PWD/brcmfmac -j2
make[1]: Entering directory '/home/aaron/Development/Github/bcm-rpi3/kernel'
  LD      /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/built-in.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/cfg80211.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/chip.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/fwil.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/fweh.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/fwsignal.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/p2p.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/proto.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/common.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/core.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/firmware.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/feature.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/btcoex.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/vendor.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/bcdc.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/sdio.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/bcmsdh.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/usb.o
  CC [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/of.o
  LD [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/brcmfmac.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/brcmfmac.mod.o
  LD [M]  /home/aaron/Development/Github/bcm-rpi3/firmware_patching/monitor_mode/brcmfmac/brcmfmac.ko
make[1]: Leaving directory '/home/aaron/Development/Github/bcm-rpi3/kernel'
../../buildtools/arm-eabi-4.7/bin/arm-eabi-gcc -Wall -Werror -O2 -nostdlib -nostartfiles -ffreestanding -mthumb -march=armv7-m -ffunction-sections -fdata-sections -c patch.c -o patch.o
patch.c: In function 'inject_frame':
patch.c:178:5: error: implicit declaration of function 'wlc_bsscfg_find_by_wlcif' [-Werror=implicit-function-declaration]
patch.c:178:12: error: assignment makes pointer from integer without a cast [-Werror]
patch.c:181:5: error: implicit declaration of function 'wlc_sendctl' [-Werror=implicit-function-declaration]
cc1: all warnings being treated as errors
Makefile:16: recipe for target 'patch.o' failed
make: *** [patch.o] Error 1
```

DanielAW, 2016-10-07:
Can you try again with 5e26db21934d6a02236ffe797678f6931cc19fbe ?

aagallag, 2016-10-07:
Thanks for your really quick response! Unfortunately I won't be able to test until tonight (8 hours or so). I'll be sure to update once I do.

aagallag, 2016-10-08:
Looks like the compile completes now.  Thanks!

closed 2016-10-08 by aagallag