seemoo-lab/bcm-rpi3 #11: What is the probabilities to damage the wifi chip?

alex9099, 2016-10-15:
Not really a bug but...
> Our software may damage your hardware and may void your hardware’s warranty! You use our tools at your own risk and responsibility! If you don't like these terms, don't use nexmon!

What is the possibility of this to happen?

matthiasseemoo, 2016-10-15:
> Not really a bug but...
>>
>> Our software may damage your hardware and may void your hardware’s
warranty! You use our tools at your own risk and responsibility! If you
don't like these terms, don't use nexmon!
>
> What is the possibility of this to happen?

It is likely that you void your warrenty. Regarding irreparable damage, we
can only say that during all of our experiments, we were not able to break
a device completely. We were able to fix any problems with a simple reboot
with the original firmware. Nevertheless, we give no guarantee that Nexmon
is always harmless.

Before using our firmware, you can always take a look at the source code
yourself and if you feel comfortable with what our firmware patches do,
feel free to use them. Otherwise, just do not use Nexmon. We do not take
responsibility for any damage that results of the use of our Nexmon
firmware as well as the firmware modifications that you create yourself
with our framework.

E3V3A, 2016-10-19:
Maybe add that to the README so you can close this issue?

alex9099, 2016-10-19:
actually it is a good idea :)

DanielAW, 2016-10-27:
The current warning in the README should me enough IMO

closed 2016-10-27 by DanielAW