seemoo-lab/bcm-rpi3 #12: How can I use this firmware into Kali Linux 2.1.2 for pi3?

caio-ribeiro-pereira, 2016-10-20:
How can I use this firmware into Kali Linux 2.1.2 for pi3?

XePeleato, 2016-10-20:
It's clearly specified in the Readme, is there any step that you don't understand?

caio-ribeiro-pereira, 2016-10-20:
@XePeleato do I need to follow only this step? 
https://github.com/seemoo-lab/bcm-rpi3#steps-to-create-your-own-firmware-patches

XePeleato, 2016-10-20:
@caio-ribeiro-pereira That's it! That will trigger the module compilation.

caio-ribeiro-pereira, 2016-10-20:
So do I need to compile on x86 hardware and follow only these steps?

+ source setup_env.sh
+ for the monitor mode patch: cd firmware_patching/monitor_mode/
+ make
+ copy the brcmfmac/brcmfmac.ko to the /root/ directory of your Raspberry Pi
+ copy the brcmfmac43430-sdio.bin to /lib/firmware/brcm/ directory of your Raspberry Pi

caio-ribeiro-pereira, 2016-10-20:
Or can I just copy and overwrite the `brcmfmac43430-sdio.bin` into `/lib/firmware/brcm` of my pi3?

XePeleato, 2016-10-20:
@caio-ribeiro-pereira You can't, because `brcmfmac43430-sdio.bin` is the patched firmware, by typing `make` the module gets compiled and the firmware gets patched, so `brcmfmac.ko` and `brcmfmac43430-sdio.bin` don't exist until you build them, the firmware present here is unpatched.

Of course, you can always download the prebuilt Raspbian image and use it.

caio-ribeiro-pereira, 2016-10-20:
And what about the unpatched firmware (https://github.com/seemoo-lab/bcm-rpi3/releases/download/0.3/brcmfmac43430-sdio.bin)? Could I just copy into my kali linux pi3?

XePeleato, 2016-10-20:
@caio-ribeiro-pereira Oh well, that firmware is patched, sorry for the misunderstanding. But still, you'd need the kernel module (`brcmfmac.ko`) to make it work.

caio-ribeiro-pereira, 2016-10-20:
@XePeleato so I just need to compile (in my case via MacOSX) to generate the `brcmfmac.ko` and copy both files to my current kali linux system? Well I'm backing up my system to test it, I hope my pi3 can works in monitor mode, thanks!

XePeleato, 2016-10-20:
@caio-ribeiro-pereira That's it! It would be nice to be able to compile the module against the current kernel, that way it would be much easier for everyone, but I'd wait until it's officially supported because you'd have to modify the Makefile and download the headers etc...

Jarvis785, 2016-10-28:
## i have a doubt !! 
### the question may be stupid for you :P
can't we get pre-compiled  **brcmfmac.ko** and **brcmfmac43430-sdio.bin** from u guys and just patch it ....  whats the reason for building ??

XePeleato, 2016-10-28:
@Jarvis785 That isn't possible, because if the module's vermagic doesn't match the kernel version, it will refuse to load it.

Jarvis785, 2016-10-28:
i didn't understand _module's vermagic doesn't match the kernel version_
but understood that it won't work ..... okey

Jarvis785, 2016-10-29:
BDW iwas downloading latest Zip of https://github.com/seemoo-lab/bcm-rpi3

but its goes on downloading without saying the file size !! 
so at 500 mb i had cancelled it ...after cancelling IDM showed downloading complete dilouge !
extracting that file gives no problem but " make" giving errors !! what to do ??
_### please help !!_

Jarvis785, 2016-10-29:
copy the image to your SD card, e.g.: dd if=nexmon_rpi_12-10-2016.dd of=/dev/mmcblk0 bs=1M note: the sd card device (in this example mmcblk0) may change depending on your setup

**how windows users will write dd to sccard ??**

SurfSound, 2016-11-20:
Hi.
I'm also at Kali Linux 2.1.2.
I followed the steps in the readme (https://github.com/seemoo-lab/bcm-rpi3#steps-to-create-your-own-firmware-patches) but I get several errors.
First of all I can't find the raspberrypi-kernel-headers. I've installed other Linux-kernel-headers instead but I don't know if this works too.
The second thing is when I try to use `make` I get the following error:

`root@kali:~/Desktop/bcm-rpi3/firmware_patching/nexmon# make
make -C ../../buildtools/gcc-nexmon-plugin/
make[1]: Entering directory '/root/Desktop/bcm-rpi3/buildtools/gcc-nexmon-plugin'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/root/Desktop/bcm-rpi3/buildtools/gcc-nexmon-plugin'
echo "Building on RPI3"
Building on RPI3
make -C /lib/modules/4.1.19-v7/build M=$PWD/brcmfmac  -j2
make[1]: *** /lib/modules/4.1.19-v7/build: No such file or directory.  Stop.
Makefile:32: recipe for target 'brcmfmac/brcmfmac.ko' failed
make: *** [brcmfmac/brcmfmac.ko] Error 2`

Does this happen because of the missing raspberrypi-kernel-headers?
And if so how can I get these?

DanielAW, 2016-11-20:
Hi,
I just learned recently about there is a Kali Linux version which runs directly on the RPI :sweat_smile:. It seems that a few people are interested in using NexMon on Kali Linux. We need to test this ourselfs, but I'm currently unsure how fast this will hapen, but we can get it on the TODO list :-)

Jarvis785, 2016-11-20:
:)

SurfSound, 2016-11-20:
yeah this would be great :) 
because my external wlan-card doesn't work properly with the latest kali linux (can't turn on monitor mode) so it would be the best if the internal would work :)
thank you

Jarvis785, 2016-11-21:
@Daniel
If nexmon works on kali..
this time make a kali img file less than 7.9 gb please..
Thanks un advance..

LilaQ, 2016-11-30:
To be honest, I really want / need Nexmon to work with Kali, since it has all the tools that would really take advantage of monitor mode.

I tried to compile it on Kali on RPi, but did not succeed so far. 
First problem you come across, Kali only supports the latest headers. Kali RPi comes with Kernel 4.1.19-v7. Kernel-Headers for this Kernel are on the image, but are not linked properly. That is why @SurfSound got his error message.

To get rid of that, remove the symlinks:

/lib/modules/4.1.19-v7/build
/lib/modules/4.1.19-v7/source
(these point to something like /home/steev/new/kali...)

Then, create both symlinks new (build and source), both pointing to /usr/src/kernel

So, now "make" will find the fitting kernel-headers, but will error with:

`linux/compiler-gcc6.h: No such file or directory`

I tried to switch to GCC-5, but only got more errors (undeclared functions etc.)

I would really love to see Nexmon working on Kali. If you guys have any hints on how to accomplish this any further, I'd be really thankful.

nethunteros, 2016-12-10:
I have been able to get Kali and Nexmon running together on my PI3.  If you are interested in the build script and repo it is here:

https://github.com/nethunteros/rpi3-kalimon

It takes a few things to get working.  I modified the [Kali Arm Build Script](https://github.com/offensive-security/kali-arm-build-scripts) for the chroot.  I use [re4son's kernel](https://github.com/nethunteros/re4son-raspberrypi-linux/tree/rpi-4.1.y-re4son) which contains support for Adafruit LCD screens also and was also preconfigured for Kali support (defconfig and mac injection).  I then applied the brcmfmac folder from Nexmon's kernel [to the new kernel](https://github.com/nethunteros/re4son-raspberrypi-linux/commit/8d24798e8ee3d22db7cec5c05bee15de80fc1269).

I forked this repo and removed the older kernel.  I added the new kernel into this repo as a submodule and finally added this repo as submodule in my rpi3-kalimon.  Finally, when the build is kicked off it will go through the normal firmware patching and kernel building using this repos makefile/patching.

Inside the final image is two files in /usr/bin called monstart and monstop.  You can easily stop and start the monitor interface by running each.

Finally, some of you probably just want the image, which for now, can be downloaded here:

https://build.nethunter.com/rpi3-nexmon/

The login/password is the usual: root | toor

_I should note I'm the only one to test and it worked fine for me._  

Eventually I would like to add support for the 3.5" interface to give a menu to quickly stop and start monitor as well as other apps like kismet or airodump.  If you would like to build your own you can use the rpi3-kalimon repo.

Jarvis785, 2016-12-11:
Great Job Buddy :) ... But the img downloading is getting failed

SurfSound, 2016-12-18:
Great work @nethunteros 

today I tested your image and it works absolutely perfect. Thank you :)
But like @Jarvis785 the download failed several times. Fortunately the download can be resumed but I had to click like 20 times on the resume button :)

cptwinder, 2016-12-19:
@nethunteros 

May I ask, what host architecture and OS are you using to run your build scripts?

I have tried directly on the RPI3 running Kali and on a Kali Light x64 VM, neither of them make the working kernel module for the wifi. I don't have in front the actual issues, or I would post them here. It seems easier for me to reproduce your build environment, then troubleshooting mine.

Thank you in advance!

denismcapple, 2016-12-21:
@nethunteros 

Thanks for your Image, it works great! I have enabled monitor mode using /usr/bin/monstart

I am having difficulty though with airodump-ng as it cannot seem to identify the channel in use

**CH 11 ][ Elapsed: 6 s ][ 2016-12-21 14:29 ][ fixed channel wlan0: -1**

There is alot of people talking about this, and the solution in most cases seems to be a driver update. There is also maxim's patch talked about here (https://www.aircrack-ng.org/doku.php?id=compat-wireless) 

Do you think this patch could be applied to this driver somehow?

said50, 2016-12-26:
Hello, I tried the kali img with nexmon but I also have download issues. So I switched back to Raspbian and I'm stuck at "set the environment in the repository folder: source setup_env.sh"
Any help?

eyfos, 2017-01-03:
@nethunteros Hi, I tried your image. It was working fine until I tried a monstart/monstop, since then no wifi, any ideas ?

DanielAW, 2017-01-09:
Closed for now, migrated to: https://github.com/seemoo-lab/nexmon/issues/20

closed 2017-01-09 by DanielAW