seemoo-lab/bcm-rpi3 #14: Segmentation fault with patched version airbase-ng

cnrat, 2016-10-23:
> GNU gdb (Raspbian 7.7.1+dfsg-5) 7.7.1
> Copyright (C) 2014 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> and "show warranty" for details.
> This GDB was configured as "arm-linux-gnueabihf".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from airbase-ng...(no debugging symbols found)...done.
> [New LWP 1670]
> [New LWP 1651]
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
> Core was generated by `airbase-ng -P -C 30 -v wlan0'.
> Program terminated with signal SIGSEGV, Segmentation fault.
> #0  memcpy () at ../ports/sysdeps/arm/memcpy.S:232
> 232     ../ports/sysdeps/arm/memcpy.S: No such file or directory.

Please check this.

DanielAW, 2016-10-24:
Hi @cnrat,
I was able to reproduce this error. In my case it runs fine for a wihile but than crashes. It seems likely that this is neither a problem of the brcmfmac driver nor our patched firmware:
* A firmware crash would produce some output in the console of the WiFI chip, which could be optained via: `cat /proc/nexmon_consoledump`. 
* A driver problem should produce some entry in the kernel log which is also not the case (at least for me)
It is more likely that this is a problem of the airodump version we compiled. I do not think that our patch causes the problems, it just tells the aircrack tool that the interface is always in monitor mode:
```
--- aircrack-ng-1.2-0~beta3.orig/src/osdep/linux.c
+++ aircrack-ng-1.2-0~beta3/src/osdep/linux.c
@@ -1517,6 +1517,10 @@ static int openraw(struct priv_linux *de
          * so just assume its correctly set...                     */
         wrq.u.mode = IW_MODE_MONITOR;
     }
+    /* NEXMON */
+    else {
+       wrq.u.mode = IW_MODE_MONITOR;
+    }

     if( ( ifr.ifr_hwaddr.sa_family != ARPHRD_IEEE80211 &&
           ifr.ifr_hwaddr.sa_family != ARPHRD_IEEE80211_PRISM &&
```
Were you able to get more information form a gdb backtrace? Recompiling the aircrack suite with debug flags enabled could also help in this case.

DanielAW, 2016-10-27:
I tested it with an unmodified version of airbase-ng using a TPLink TL-WN722N WiFi stick on the RaspberryPi3. It still SegFaults for me after a while. It seems to be a bug in aircrack-ng.

Also: Using our current modified driver does not require the aircrack-ng patch anymore.

closed 2016-10-27 by DanielAW