seemoo-lab/bcm-rpi3 #15: Suggestion: binary patch brcmfmac.ko to support any new kernel

E3V3A, 2016-10-23:
Unfortunately the default Rasbian kernel is configured with `MODULE_FORCE_LOAD` **disabled**, thus preventing a force module insertion via `sudo modprobe --force-vermagic brcmfmac`.

```bash
sudo modprobe configs
zgrep MODULE_FORCE_LOAD /proc/config.gz
# CONFIG_MODULE_FORCE_LOAD is not set
sudo rmmod configs
```

**So what to do?** Well, we could try the ugly hack of binary patching any recently built patched bcm-rpi3 module, by replacing the `vermagic=...` with what you find from `uname -a`. For example, do the equivalent for patched .ko like this:

```bash
cd /lib/modules/4.4.21-v7+/kernel/drivers/net/wireless/brcm80211/brcmfmac/
strings -a -t x brcmfmac.ko | grep -E "4.4.21-v7"

  206c3 vermagic=4.4.21-v7+ SMP mod_unload modversions ARMv7

# "vermagic=" --> +9 char. ==> c3+9 = cc
hexdump -C -n32 -s0x206cc brcmfmac.ko

000206cc  34 2e 34 2e 32 31 2d 76  37 2b 20 53 4d 50 20 6d  |4.4.21-v7+ SMP m|
000206dc  6f 64 5f 75 6e 6c 6f 61  64 20 6d 6f 64 76 65 72  |od_unload modver|

```
(I'll let you do the binary editing as a homework assignment.) 

Why do all this ugly hacks? To avoid having to cross-compile the entire kernel to get injection working. Sure, great academic exercise but a lots of wasted time. 

@DanielAW Are the modules also version or CRC checked?

matthiasseemoo, 2016-10-23:
That is another hack that should be avoided. The best option would be to integrate our monitor mode patch into the main brcmfmac repository and thereby make it the default kernel module for the raspberry pi 3. On the Nexus 5 we already have a firmware version that works for regular wifi operation and can enable monitor mode on demand. it would make sense to get to a similar state for the raspberry pi 3 but also write some code to offer a clean monitor mode implementation in the brcmfmac driver. As we do not have the time to dive into the driver, it would be good to find someone who knows the driver well and can support us with this task.

DanielAW, 2016-10-27:
We now support building the fw patches and driver on the rpi3 itself. Building the kernel is not required anymore => closed

closed 2016-10-27 by DanielAW