seemoo-lab/bcm-rpi3 #17: Missing monitor_mode/-directory

ejric, 2016-10-24:
Hello!

Im on **Kali Linux 2.1.2** and if I understand this right, I just need to download the files and follow the steps in "**_Steps to create your own firmware patches_**"...?

So I downloaded it to a USB and plugged in to my RPI3.

I cd into the folder where "**setup_env.sh**" is located and execute the command "**source setup.....**",
next step is to "**cd monitor_mode/**", and that folder is missing under the "**firmware_patching**"-directory.

I'm a bit new to linux so this might be why it doesn't work... layer 8-problems.. 

Glad if u can help me out!

Thanks!
ejriC

XePeleato, 2016-10-24:
Hi, the folder has been renamed to 'nexmon'. 

I hope that solves your problem.

ejric, 2016-10-24:
Found it!

But still, cannot get it to work.

Did the "**brcmfmac/brcmfmac.ko**" changed location aswell, cause can't find it..

And it should just be copying this files into the locations and it will work? Or do i have to remove the old firmware?

XePeleato, 2016-10-24:
Yes, it's still at the same folder, maybe the compilation didn't complete. Would you mind to post the log? I can check it out. 

And yes, you just need to copy them.

ejric, 2016-10-24:
`root@kali:/media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon# make
make -C ../../kernel/ M=$PWD/brcmfmac -j2
make[1]: Entering directory '/media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/kernel'
make[1]: execvp: /media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: Permission denied

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  LD      /media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac/built-in.o
/bin/sh: 1: /media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-ar: Permission denied
scripts/Makefile.build:337: recipe for target '/media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac/built-in.o' failed
make[2]: *** [/media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac/built-in.o] Error 126
make[2]: *** Waiting for unfinished jobs....
Makefile:1385: recipe for target '_module_/media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac' failed
make[1]: *** [_module_/media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac] Error 2
make[1]: Leaving directory '/media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/kernel'
Makefile:29: recipe for target 'brcmfmac/brcmfmac.ko' failed
make: *** [brcmfmac/brcmfmac.ko] Error 2
root@kali:/media/root/ERIC/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon# 
`

There you go!

XePeleato, 2016-10-24:
As the log suggests `Run 'make oldconfig && make prepare' on kernel src to fix it. ` it used to work but someone commented a couple of lines and now it drops that warning. Hint: (The kernel source is at the kernel folder.)

ejric, 2016-10-24:
did that and got the "ERROR" away but still the "WARNING.

`root@kali:~/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon# make
make -C ../../kernel/ M=$PWD/brcmfmac -j2
make[1]: Entering directory '/root/bcm-rpi3-master/bcm-rpi3-master/kernel'
make[1]: execvp: /root/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: Permission denied

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  LD      /root/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac/built-in.o
/bin/sh: 1: /root/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-ar: Permission denied
scripts/Makefile.build:337: recipe for target '/root/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac/built-in.o' failed
make[2]: *** [/root/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac/built-in.o] Error 126
Makefile:1385: recipe for target '_module_/root/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac' failed
make[1]: *** [_module_/root/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon/brcmfmac] Error 2
make[1]: Leaving directory '/root/bcm-rpi3-master/bcm-rpi3-master/kernel'
Makefile:29: recipe for target 'brcmfmac/brcmfmac.ko' failed
make: *** [brcmfmac/brcmfmac.ko] Error 2
`

Seems like i'm missing some permissions?

XePeleato, 2016-10-24:
It used to work, but this commit: cd48f25 disabled the two lines that used to build the full kernel, maybe @DanielAW has an answer since I really don't know if there's any alternative, besides building the whole kernel. I suggested #16 in order to avoid this kind of problems.

DanielAW, 2016-10-24:
> It used to work, but this commit: d5129fb disabled the two lines that used to build the full kernel

Woops, my bad. This was _not_ intended :-) Should be fixed with 6669f45

ejric, 2016-10-24:
Hi @DanielAW !

I replaced the **Make-file** in  "**firmware_patching/nexmon/**" with [6669f45](https://github.com/seemoo-lab/bcm-rpi3/commit/6669f4570c8d585b54d330e953428fee0c7fa019) and then tried to run "**Make**" again from "**firmware_patching/nexmon/**" but i still get the same warnings. 

Get failed to create "**brcmfmac.ko**"

Is some other steps required then just replacing the **Make-file** in  "**firmware_patching/nexmon/**"?

XePeleato, 2016-10-24:
@ejric Have you tried to give the proper permissions to the files that fail to run? Just run `chmod +x {path} ` with proper permissions.

ejric, 2016-10-25:
Yes I've done that. Here's the log:

```
root@kali:~/bcm-rpi3-master/bcm-rpi3-master/firmware_patching/nexmon# make 
make -C ../../kernel/ bcm2709_defconfig
make[1]: Entering directory '/root/bcm-rpi3-master/bcm-rpi3-master/kernel'
#
# configuration written to .config
#
make[1]: Leaving directory '/root/bcm-rpi3-master/bcm-rpi3-master/kernel'
make -C ../../kernel/ -j2
make[1]: Entering directory '/root/bcm-rpi3-master/bcm-rpi3-master/kernel'
/root/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: /root/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: arm-linux-gnueabihf-gcc-4.8.3: not found
scripts/kconfig/conf  --silentoldconfig Kconfig
/root/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: /root/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: arm-linux-gnueabihf-gcc-4.8.3: not found
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CC      scripts/mod/empty.o
/root/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: 1: /root/bcm-rpi3-master/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: arm-linux-gnueabihf-gcc-4.8.3: not found
scripts/Makefile.build:258: recipe for target 'scripts/mod/empty.o' failed
make[3]: *** [scripts/mod/empty.o] Error 127
scripts/Makefile.build:403: recipe for target 'scripts/mod' failed
make[2]: *** [scripts/mod] Error 2
Makefile:545: recipe for target 'scripts' failed
make[1]: *** [scripts] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/root/bcm-rpi3-master/bcm-rpi3-master/kernel'
Makefile:27: recipe for target 'brcmfmac/brcmfmac.ko' failed
make: *** [brcmfmac/brcmfmac.ko] Error 2
```
I've have checked, and all the files that get "**Not found:**" is NOT missing, they are in the right folders and I have -rw - permissons to all.

E3V3A, 2016-10-25:
This is your hint:
```
arm-linux-gnueabihf-gcc-4.8.3: not found
```

XePeleato, 2016-10-25:
Oh wait, you are trying to compile it with your Rpi, that's not possible at the moment, you'll have to use a computer or wait until Rpi support is included.

MasterfulMethods, 2016-10-25:
Hi, looking forward to testing this project, however, I'm having the same issue as @ejric 
I AM attempting to compile from x86. 

uname -a: Linux 4.4.0-38-generic #57-Ubuntu SMP x86_64 GNU/Linux

```
nightfire@nightfire-lx:~/Downloads/bcm-rpi3-master/firmware_patching/nexmon$ make
make -C ../../kernel/ bcm2709_defconfig
make[1]: Entering directory '/home/nightfire/Downloads/bcm-rpi3-master/kernel'
#
# configuration written to .config
#
make[1]: Leaving directory '/home/nightfire/Downloads/bcm-rpi3-master/kernel'
make -C ../../kernel/ -j2
make[1]: Entering directory '/home/nightfire/Downloads/bcm-rpi3-master/kernel'
make[1]: /home/nightfire/Downloads/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: Command not found
scripts/kconfig/conf  --silentoldconfig Kconfig
make[1]: /home/nightfire/Downloads/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: Command not found
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CC      scripts/mod/empty.o
/bin/sh: 1: /home/nightfire/Downloads/bcm-rpi3-master/buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc: not found
scripts/Makefile.build:258: recipe for target 'scripts/mod/empty.o' failed
make[3]: *** [scripts/mod/empty.o] Error 127
scripts/Makefile.build:403: recipe for target 'scripts/mod' failed
make[2]: *** [scripts/mod] Error 2
Makefile:545: recipe for target 'scripts' failed
make[1]: *** [scripts] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/nightfire/Downloads/bcm-rpi3-master/kernel'
Makefile:27: recipe for target 'brcmfmac/brcmfmac.ko' failed
make: *** [brcmfmac/brcmfmac.ko] Error 2
```

Also, i've attempted to dd the ready-made image twice and I get about 1/2 through boot and get:
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2)

```
dd if=nexmon_rpi_12-10-2016.dd of=/dev/sde bs=1M status=progress
```

Thoughts?

Thanks

DanielAW, 2016-10-26:
Seems like most people struggle with compiling the linux kernel. I hope we can circumvent this issue by finishing #18. 
@MasterfulMethods this is strange, if the file is there it can only be a problem with the platform its build for, but this should be fine in your case. You can verify it via: `file buildtools/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-gcc-4.8.3`
Regarding the dd image: let me test this image myself, I may got a faulty SD Card.

DanielAW, 2016-10-26:
@MasterfulMethods the dd image runs fine for me, here are the md5sums:
```
af2c2df3df547025dd01e48bdb163322  nexmon_rpi_12-10-2016.dd
82ff4d7c9e7980d0568f972f728e8544  nexmon_rpi_12-10-2016.tar.gz
```

MasterfulMethods, 2016-10-26:
Thanks @DanielAW 

I'll redownload later today and verify the checksums on my end

XePeleato, 2016-10-26:
I have downloaded the source again and it builds fine here.

Maybe you are missing bc? `apt install bc`. I assume GCC is installed.

ejric, 2016-10-27:
Redownloaded clone and installed bc.

Did the apt-get update/upgrade w/o errors.

Next step is where i'm stuck.

```
root@kali:~# sudo apt install raspberrypi-kernel-headers libgmp3-dev gawk
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package raspberrypi-kernel-headers
```
`apt search` doesn't show _"raspberrypi-kernel-headers"_
Either do`apt find`

Am I missing something here? :p

ejric, 2016-10-27:
Nvm!

Forgot to do the `git clone https://github.com/seemoo-lab/bcm-rpi3.git`-command.
I'll test that right away!

DanielAW, 2016-10-27:
I just saw: `root@kali`
[These](https://github.com/seemoo-lab/bcm-rpi3#using-your-raspberry-pi) steps are supposed to be executed on the RPI3 itself. You have to use raspbian or something like that.

Jarvis785, 2016-10-28:
i think only theses steps are needed 
 just building using a linux computer and then copy to raspberry pi 3
https://github.com/seemoo-lab/bcm-rpi3#using-a-x86-based-computer
and after these step just reboot or what ???

these steps are not required ???
https://github.com/seemoo-lab/bcm-rpi3#using-your-raspberry-pi

am i right ??

Jarvis785, 2016-10-28:
BDW i'm now building using Linux Mint ....

**how long it will take... any ideas ??**

XePeleato, 2016-10-28:
@Jarvis785 Have you loaded the module? `insmod brcmfmac.ko` Have you tried tcpdump to check if it's in monitor mode?

Jarvis785, 2016-10-28:
wait....

ejric, 2016-10-28:
@Jarvis785 How are you trying to access your WIFI?
Try editing the **interfaces**-file `nano /etc/network/interfaces`

add the following to your file:
`auto wlan0`
`iface wlan0 inet dhcp`
`wpa-ssid “your network name”`
`wpa-psk “the network password”`

Interface (wlan0) may vary depending on ypur device.
_Ctrl+X -> Y(es) -> Enter_

Jarvis785, 2016-10-28:
ohh got it 
i think after patching...

 _/etc/wpa_supplicant/wpa_supplicant.conf_ won't work ??

ejric, 2016-10-28:
Ye I guess! Did it work? Don't know if u need to change the `wpa-ssid & wpa-psk` depending on wich protocol you using. For example `wpa2-ssid...`

And also reboot after that was needed for me.

Hope that helps!

Jarvis785, 2016-10-28:
[Should i do this ?](https://github.com/seemoo-lab/bcm-rpi3/blob/master/README.md#using-your-raspberry-pi)

ejric, 2016-10-28:
If I've  got this right (_I'm kinda NOOB to_),

If you are on your RPI:
Follow **Using your Raspberry Pi**-steps.. ONLY!

If you are on a x86 (32bit) computer:
Follow **Using a x86 based computer**-steps ONLY!

And for some OS's this won't work, for example **KALI** which I'm on.

But if you are on Raspbian OS this should work.

Jarvis785, 2016-10-28:
haha..

ejric, 2016-10-28:
The `make` - command is creating the **brcmfmac.ko**-file

Jarvis785, 2016-10-28:
## i know that..

Jarvis785, 2016-10-29:
### @ejric  this is not working .... :(
> @Jarvis785 How are you trying to access your WIFI?
> Try editing the interfaces-file nano /etc/network/interfaces
>
> add the following to your file:
> auto wlan0
> iface wlan0 inet dhcp
> wpa-ssid “your [...]

ejric, 2016-10-29:
Type `ifconfig -a`, is your WIFI-card showing up there? (wlan0). If yes:

In `nano /etc/network/interfaces`, try wpa-ssid/wpa-psk without "". Then reboot and try again.

PS: Everything in Linux is CasEsEnSiTiVe

Jarvis785, 2016-10-29:
## this worked !! :com
**auto lo
iface lo inet loopback
auto eth0
allow-hotplug eth0
iface eth0 inet dhcp
auto wlan0
iface wlan0 inet dhcp
wpa-ssid "JARVIS"
wpa-psk "12345678"**

ejric, 2016-10-29:
Zipped: 545 701 kB

Would you mind posting the log?

Jarvis785, 2016-10-29:
i have compiled in linux and copied...
getting this error
**root@Pi-Jarvis:/home/pi# insmod /home/pi/brcmfmac.ko
insmod: ERROR: could not insert module /home/pi/brcmfmac.ko: Invalid module format**

ejric, 2016-10-29:
@Jarvis785 Hmm.. seems like there is a  mismatch between your kernel version and header files.

Maybe try the `sudo apt install raspberrypi-kernel-headers libgmp3-dev gawk`-command anyway (if you havn't done it).

Or what do you guys say? @XePeleato @DanielAW

Jarvis785, 2016-10-29:
no i have done it... **sudo apt install raspberrypi-kernel-headers libgmp3-dev gawk**
@ejric

DanielAW, 2016-10-29:
>  insmod: ERROR: could not insert module /home/pi/brcmfmac.ko: Invalid module format
It seems likely that the version magic of the module does not fit the kernel version number. Can you show me the output of `modinfo /home/pi/brcmfmac.ko` and `uname -a` (on the RPI3)?

Jarvis785, 2016-10-29:
here is it ..
**pi@Pi-Jarvis:~ $ modinfo /home/pi/brcmfmac.ko
filename:       /home/pi/brcmfmac.ko
license:        Dual BSD/GPL
description:    Broadcom 802.11 wireless LAN fullmac driver.
author:         Broadcom Corporation
firmware:       brcm/brcmfmac4354-sdio.txt
firmware:       brcm/brcmfmac4354-sdio.bin
firmware:       brcm/brcmfmac43455-sdio.txt
firmware:       brcm/brcmfmac43455-sdio.bin
firmware:       brcm/brcmfmac43430-sdio.txt
firmware:       brcm/brcmfmac43430-sdio.bin
firmware:       brcm/brcmfmac4339-sdio.txt
firmware:       brcm/brcmfmac4339-sdio.bin
firmware:       brcm/brcmfmac43362-sdio.txt
firmware:       brcm/brcmfmac43362-sdio.bin
firmware:       brcm/brcmfmac4335-sdio.txt
firmware:       brcm/brcmfmac4335-sdio.bin
firmware:       brcm/brcmfmac43341-sdio.txt
firmware:       brcm/brcmfmac43341-sdio.bin
firmware:       brcm/brcmfmac43340-sdio.txt
firmware:       brcm/brcmfmac43340-sdio.bin
firmware:       brcm/brcmfmac4334-sdio.txt
firmware:       brcm/brcmfmac4334-sdio.bin
firmware:       brcm/brcmfmac4330-sdio.txt
firmware:       brcm/brcmfmac4330-sdio.bin
firmware:       brcm/brcmfmac4329-sdio.txt
firmware:       brcm/brcmfmac4329-sdio.bin
firmware:       brcm/brcmfmac43241b5-sdio.txt
firmware:       brcm/brcmfmac43241b5-sdio.bin
firmware:       brcm/brcmfmac43241b4-sdio.txt
firmware:       brcm/brcmfmac43241b4-sdio.bin
firmware:       brcm/brcmfmac43241b0-sdio.txt
firmware:       brcm/brcmfmac43241b0-sdio.bin
firmware:       brcm/brcmfmac43143-sdio.txt
firmware:       brcm/brcmfmac43143-sdio.bin
firmware:       brcm/brcmfmac43569.bin
firmware:       brcm/brcmfmac43242a.bin
firmware:       brcm/brcmfmac43236b.bin
firmware:       brcm/brcmfmac43143.bin
srcversion:     54FC7DE8E57FF8CECA23D5F
alias:          usb:v0A5Cp0BDCd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD27d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD1Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD17d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD1Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          sdio:c*v02D0d4354*
alias:          sdio:c*v02D0d4345*
alias:          sdio:c*v02D0dA9A6*
alias:          sdio:c*v02D0d4335*
alias:          sdio:c*v02D0dA962*
alias:          sdio:c*v02D0dA94D*
alias:          sdio:c*v02D0dA94C*
alias:          sdio:c*v02D0d4334*
alias:          sdio:c*v02D0d4330*
alias:          sdio:c*v02D0d4329*
alias:          sdio:c*v02D0d4324*
alias:          sdio:c*v02D0dA887*
depends:        brcmutil,cfg80211
vermagic:       4.4.13-v7 SMP mod_unload modversions ARMv7
parm:           txglomsz:maximum tx packet chain size [SDIO] (int)
parm:           feature_disable:Disable features (int)
parm:           alternative_fw_path:string
parm:           debug:level of debug output (int)
parm:           p2pon:enable legacy p2p management functionality (int)
parm:           fcmode:mode of firmware signalled flow control (int)
parm:           roamoff:do not use internal roaming engine (int)**

**### pi@Pi-Jarvis:~ $ uname -a
Linux Pi-Jarvis 4.4.26-v7+ #915 SMP Thu Oct 20 17:08:44 BST 2016 armv7l GNU/Linux**

DanielAW, 2016-10-29:
The kernel on your Raspberry Pi 3 (**4.4.26-v7**) is newer than the kernel which you use to build the kernel module (**4.4.13-v7**). I think the easiest way to solve this would be compiling the driver on the Raspberry Pi itself.

Jarvis785, 2016-10-29:
actually..
i have downloaded  https://github.com/seemoo-lab/bcm-rpi3
the latest one but... its gets downloaded and downloded IDM doesn't shows its size neither % ... keeps downloading .... when size became 500+ mb i stooped it showed download complete ... i copied that to raspberry and compiled ... got some erroes...

i hve a zip old version if bcm-rpi3-0.3.zip
now i will compile it and report u if its working or not..
thanks @DanielAW 
# :)

Jarvis785, 2016-10-29:
No
 while compiling on RP3 its giving this error .....
please say what to do
@DanielAW 
[error.txt](https://github.com/seemoo-lab/bcm-rpi3/files/560005/error.txt)

DanielAW, 2016-10-29:
Thats (sort of) good :-) It means that the kernel module was successfully compiled. But it seems there is something wrong with our kernel add-on. Can you check if there is a `nexmon.so` in this folder?: `/media/pi/F8720A9A720A5DAE/bcm-rpi3-master/buildtools/gcc-nexmon-plugin`

Jarvis785, 2016-10-29:
let me check

Jarvis785, 2016-10-29:
yes nexmo.so is there

DanielAW, 2016-10-29:
Ok, so it successfully compiled. What does this command, executed from the "nexmon" folder return? `../../buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc -fplugin=../../buildtools/gcc-nexmon-plugin/nexmon.so`

Jarvis785, 2016-10-29:
wait...

Jarvis785, 2016-10-29:
pi@Pi-Jarvis:/media/pi/F8720A9A720A5DAE/bcm-rpi3-0.3/firmware_patching/monitor_mode $ ../../buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc -fplugin=../../buildtools/gcc-nexmon-plugin/nexmon.so
-bash: ../../buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc: No such file or directory

actully im using older version so nexmon is not there (monitor_mode is therer)

DanielAW, 2016-10-29:
Ok, strange. The other error messages indicate that you use a newer version including a "nexmon" folder. It seems the arm gcc is missing, I just added it recently. You will need the arm gcc for your compiler add-on. You should have a directory called `buildtools/gcc-arm-none-eabi-5_4-2016q2_arm` (starting from the repositories root directory). Maybe you need to download the latest master branch.

Jarvis785, 2016-10-29:
okey .... first say how i can download latest one ??

DanielAW, 2016-10-29:
Your previous messages indicate that you downloaded our poject as .zip right? Using git assures that you always got the latest version, just use `git clone git@github.com:seemoo-lab/bcm-rpi3.git`. Now you should be able to follow our steps described [here](https://github.com/seemoo-lab/bcm-rpi3#using-your-raspberry-pi). Starting with "set the environment in the repository folder"

Jarvis785, 2016-10-29:
shall i download usin github desktop ??

Jarvis785, 2016-10-29:
but doing .. git clone git@github.com:seemoo-lab/bcm-rpi3.git  will download 3oo mb file on raspberry ??

DanielAW, 2016-10-29:
That sounds about right, our current repo contains the RPI3 kernel and 2 versions of gcc. The console version of git should be ok.

Jarvis785, 2016-10-29:
300 mb on raspberry !! too slow shall i use github desktop ??

DanielAW, 2016-10-29:
300 MB should be no problem in my opinion. Using Github Desktop on the RPI3 will _also_ download 300 MB. Is your RPI connected via WiFi? If this is the case, please consider: You will **not** be able to use Monitor Mode and Station Mode (default mode) at the same time with NexMon. You will need to exchange the driver as well as the firmware every time you want to switch mode.

Jarvis785, 2016-10-29:
i know i can't control GUI WIFI(no wifi symbol on taskbar) but i think i can connect to a wifi network
by editing :::: sudo nano /etc/network/interfaces

auto wlan0
iface wlan0 inet dhcp
wpa-ssid "JARVIS"
wpa-psk "12345678"

what do u say ?? it work or not ???

bdw may i have ur FB or something to contact u for faster messeging...

[6 more replies]

closed 2016-10-29 by DanielAW