seemoo-lab/bcm-rpi3 #19: how to write dd to sdcard on windows ??(Win32DiskImager)

Jarvis785, 2016-10-29:
copy the image to your SD card, e.g.: dd if=nexmon_rpi_12-10-2016.dd of=/dev/mmcblk0 bs=1M note: the sd card device (in this example mmcblk0) may change depending on your setup

how windows users will write dd to sccard ??

matthiasseemoo, 2016-10-29:
by becoming a linux or mac user ;-)

Jarvis785, 2016-10-29:
okey if thats the only way......

DanielAW, 2016-10-29:
As you already try the [diskmanager](https://sourceforge.net/projects/win32diskimager/)? I thought we got another one which successfully used this tool.

Jarvis785, 2016-10-29:
what do u mean ??

DanielAW, 2016-10-29:
"win32diskimager" is more or less a subsitute for "dd" on Windows

Jarvis785, 2016-10-29:
actually win32 disk imager saying out of space (8gb card)

DanielAW, 2016-11-15:
closed, same problem as in #20

closed 2016-11-15 by DanielAW