seemoo-lab/bcm-rpi3 #20: Latest bcm-rpi3 with latest Raspberry giving make Error

Jarvis785, 2016-10-30:
@DanielAW  have a look

`pi@Pi-Jarvis:/media/pi/70DCCE9BDCCE5B4C/bcm-rpi3/firmware_patching/nexmon $ cd -
/media/pi/70DCCE9BDCCE5B4C/bcm-rpi3
pi@Pi-Jarvis:/media/pi/70DCCE9BDCCE5B4C/bcm-rpi3 $ source setup_env.sh
Raspberry Pi detected, disabling cross-compilation.
pi@Pi-Jarvis:/media/pi/70DCCE9BDCCE5B4C/bcm-rpi3 $ cd firmware_patching/nexmon/
pi@Pi-Jarvis:/media/pi/70DCCE9BDCCE5B4C/bcm-rpi3/firmware_patching/nexmon $ make
make -C ../../buildtools/gcc-nexmon-plugin/
make[1]: Entering directory '/media/pi/70DCCE9BDCCE5B4C/bcm-rpi3/buildtools/gcc-nexmon-plugin'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/media/pi/70DCCE9BDCCE5B4C/bcm-rpi3/buildtools/gcc-nexmon-plugin'
echo "Building on RPI3"
Building on RPI3
make -C /lib/modules/4.4.26-v7+/build M=$PWD/brcmfmac  -j2
make[1]: Entering directory '/usr/src/linux-headers-4.4.26-v7+'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.4.26-v7+'
../../buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc \
	-fplugin=../../buildtools/gcc-nexmon-plugin/nexmon.so \
	-fplugin-arg-nexmon-fwfile=brcmfmac43430-sdio.bin \
	-fplugin-arg-nexmon-ramstart=0x0 \
	-fplugin-arg-nexmon-objfile=patch.o \
	-fplugin-arg-nexmon-prefile=nexmon.generated.pre \
	-fplugin-arg-nexmon-chipver=`printf "#include \"../include/firmware_version.h\"\n%s\n" CHIP_VER_BCM43438 | ../../buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc -E -x c - | tail -n 1` \
	-fplugin-arg-nexmon-fwver=`printf "#include \"../include/firmware_version.h\"\n%s\n" FW_VER_7_45_41_26_r640327 | ../../buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc -E -x c - | tail -n 1` \
	-DNEXMON_CHIP=CHIP_VER_BCM43438 \
	-DNEXMON_FW_VERSION=FW_VER_7_45_41_26_r640327 \
	-Wall -Werror -O2 -nostdlib -nostartfiles -ffreestanding -mthumb -march=armv7-r -ffunction-sections -fdata-sections -I../include -c patch.c -o patch.o
/bin/sh: 1: ../../buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc: Permission denied
/bin/sh: 1: ../../buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc: Permission denied
/bin/sh: 1: ../../buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc: Permission denied
Makefile:45: recipe for target 'patch.o' failed
make: *** [patch.o] Error 126

latest `bcm-rpi3 everything done... as per instruction
why i get this error 😞  😢

DanielAW, 2016-10-30:
Both the add-on and the compiler itself are there? Can you show us the output of `file buildtools/gcc-nexmon-plugin/nexmon.so` and `file buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc` ? What output does `buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc -fplugin=buildtools/gcc-nexmon-plugin/nexmon.so` give you?

Jarvis785, 2016-10-30:
Okey i will ket u know..

Jarvis785, 2016-10-30:
in /bcm-rpi3 directory 
first entered ::: source setup_env.sh

then..
@DanielAW 

root@Pi-Jarvis:/media/pi/70DCCE9BDCCE5B4C1/bcm-rpi3# file buildtools/gcc-nexmon-plugin/nexmon.so
buildtools/gcc-nexmon-plugin/nexmon.so: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=26910e3e6df31f5fa34552d9bc2140fd81182d32, not stripped

root@Pi-Jarvis:/media/pi/70DCCE9BDCCE5B4C1/bcm-rpi3# file buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc
buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=dd0e6709a923d3897ddc863547c3429aa2c6fd60, not stripped

root@Pi-Jarvis:/media/pi/70DCCE9BDCCE5B4C1/bcm-rpi3# buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc -fplugin=buildtools/gcc-nexmon-plugin/nexmon.so
bash: buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc: Permission denied
]

DanielAW, 2016-11-02:
This might sound trivial but can you check the file permissions on `buildtools/gcc-arm-none-eabi-5_4-2016q2_arm/bin/arm-none-eabi-gcc` ?

Jarvis785, 2016-11-02:
How to check permission ? Chmod value ?

Jarvis785, 2016-11-11:
@daniel help me .... I need to make raspberry jamming device !! Hows that possible without monitor mode !!

matthiasseemoo, 2016-11-11:
We will not give any instructions on how to build jammers with the nexmon
firmware. If you need a jammer, figure it out on your own.

Jarvis785, 2016-11-11:
No .... I just unable to activate monitor mode .... If this priblem will be solved the i can jamm it

Jarvis785, 2016-11-11:
And I'm unable to load ur pre patched firmware dd image in 4gb sdcard ... It's saying out of size !!

DanielAW, 2016-11-11:
As stated in the README: you need a 8 GB sdcard for this image

Jarvis785, 2016-11-11:
Ohh sorry my bad .. I'm unable to load on 8gb its showing .... Out of space..

DanielAW, 2016-11-11:
Did you use the Win32Disk Image tool you mentioned in the other issue? This might help you: http://elinux.org/RPi_Easy_SD_Card_Setup#Flashing_the_SD_Card_using_Windows

Jarvis785, 2016-11-11:
I had flashed many rasbian images using win32 no issues ... But unable to flash your dd on 8gb

Jarvis785, 2016-11-11:
For example im getting error like following:--
Not enough space on disk: Size 31553536 sectors Available 31537152 sectors
I mean it just wants some 2-3 mb extra (7.9gb+extra) .... May be i need to shrink the dd img (because extra empty space also makes the img size big)
(Note:: my size is not 31553536 ,this for example)

Jarvis785, 2016-11-12:
I am able to resize my raspberry backup .img using gparted ... But unable to resize yor .dd format file .... Why ?? ... 

Can u give me .img backup of yor nexmon (full os)

matthiasseemoo, 2016-11-12:
if you are willing to pay a hundred bucks as a service fee, we can do this.
otherwise, just use ubuntu to install our image to an sd card.

Jarvis785, 2016-11-12:
Sorry I'm unable to afford that :(

Jarvis785, 2016-11-12:
root@kali:/media/My Passport/8 Mile# dd if=nexmon_rpi_12-10-2016.dd of=/dev/sdc bs=1M
**dd: writing `/dev/sdc': No space left on device**
7549+0 records in
7548+0 records out
7914651648 bytes (7.9 GB) copied, 1751.52 s, 4.5 MB/s

help me guys.... its a sandisk 8gb Sdcard !! why this is happening

Jarvis785, 2016-11-13:
i think nexmon_rpi_12-10-2016.dd is not working !!

i have used dd if=nexmon_rpi_12-10-2016.dd of=/dev/sdc bs=1M on a 32 gb card ..
7549+0 records in
7548+0 records out
7914651648 bytes (7.9 GB) copied, 1751.52 s, 4.6 MB/s

all dd process executed without errors..

But still raspberry not booting !! What the hell !!

help me guys..

@DanielAW @matthiasseemoo  @ophilli @schmittner @XePeleato

DanielAW, 2016-11-13:
> What the hell !!

Beeing rude won't help anybody :-1: :disappointed: Any errors on `dmesg` after/before copying the image?

Jarvis785, 2016-11-15:
Okey anyone knows how to solve this ?

Write nexmon_rpi_12-10-2016.dd  to 8gb sdcard using windows/linux ...
Because it's saying out of space in win32 disk manager as well as in linux ...

And also i had written dd file to 32 gb sdcard using linux without any errors ...but the raspberry is not booting up !!

DanielAW, 2017-01-09:
Closed for now, this repository is deprecated. Please report any problems to https://github.com/seemoo-lab/nexmon

closed 2017-01-09 by DanielAW