seemoo-lab/bcm-rpi3 #23: compiling aircrack-ng from SVN

chronos000, 2016-11-16:
@DanielAW 
@matthiasseemoo 

I have used your latest image for the raspberry pi 3 (nexmon_rpi_12-10-2016.tar.gz) and succesfully booted and played around with airodump-ng.

I was wishing that besside-ng would work just as flawlessly, but unfortunately it was not the case. I am constantly receiving all kinds of "wi_read()" errors.

I read at the aircrack-ng bug tracker that these type of errors have been fixed in the experimental branch of aircrack-ng.

This is why I tried downloading the latest aircrack-ng version from SVN and compiled it. But I am missing the patches you added so that these scripts recognize the interface in monitor mode even though it reports to be in managed mode (among other things, I guess). I can't find these patches anywhere in your repos, nor any instructions on how to do this.

Could you please tell me how to patch the latest version of the aircrack-ng suit so that it would keep working with nexmon?

Thank you

DanielAW, 2016-11-16:
Hi,
the current release for the RPI3 does not require paching aircrack-ng anymore. Our modified driver shows itself as monitor device to the operating system. 
The dd-able image which is currently available includes an older version of the driver. You can build the new driver from the repository. Using the new kernel module should make the pachted aircrack-ng version obsolete. If you plan on compiling the driver on the RPI3 itself (which works without compiling the complete kernel), you can also checkout our new work-in-progress repository here: https://github.com/seemoo-lab/nexmon

chronos000, 2016-11-18:
Excellent. Thank you. I have tried that and found problems while compiling. I'll open new issues appropriately in the nexmon repo.

closed 2016-11-18 by chronos000