seemoo-lab/bcm-rpi3 #24: Doesn't seem to be working with latest raspbian

BradHuber, 2016-11-24:
Make error likely due to path issues (and more?).

brcmfmac wants /lib/modules/4.4.21-v7+ whereas the latest version is /lib/modules/4.4.32-v7+

how to fix?

XePeleato, 2016-11-24:
@BradHuber Could you please run `uname -r` and paste the output here?

BradHuber, 2016-11-25:
uname -r gives:
4.4.32-v7+

BradHuber, 2016-11-26:
I used the latest raspbian image (4.4.21) without doing apt-get upgrade.  With this version I am able to get the firmware patch compiled and installed.  I won't worry about getting it working with the latest version for now.

Follow-on question- after installation iw-config shows that wlan0 is in monitor mode.  However within wireshark/tshark I cannot use the -I option on the commandline or via the GUI.  So am I still capturing in monitor mode?  When I start capture on wlan0 I do see packets, but I cannot get wireshark to filter on wlan.fc.type == 0 etc...

Any thoughts?

michaelcalve, 2016-11-29:
In Wireshark, are you talking about enabling monitor mode for the wlan connection?

BradHuber, 2016-11-30:
Well... yes.  But let me be more specific.

I am trying to use the Raspberry Pi to sniff WiFi probe requests.  Initially I am trying to use wireshark/tshark to do it.

Whether or not wireshark sets the adapter into monitor mode or if it is already in monitor mode doesn't matter to me.

iwconfig reports that the device is in monitor mode.. yet I can't seem to sniff the probe requests (none appear in wireshark).

BradHuber, 2016-11-30:
And yes I am talking about the wlan0 adapter on the ras-pi

BradHuber, 2016-11-30:
It's the built in wlan interface on the raspbery pi 3

michaelcalve, 2016-11-30:
Hmmm. Im actually trying to do the same thing since wireshark cant enable
monitoring on my pi

DanielAW, 2016-11-30:
@BradHuber You should be able to use Wireshark without using the `-I` option. The Wifi interface is _always_ in monitor mode if you use our patched driver. Apart from the driver it is also essential to replace the original firmware in `/lib/firmware/brcm`. Does tcpdump work? (e.g.: `tcpdump -i wlan0 -s0`)

DanielAW, 2016-12-05:
Closed for now, no new infos.

closed 2016-12-05 by DanielAW