seemoo-lab/bcm-rpi3 #25: Not able to load up the tar.gz image file on to my raspberry pi from my SD

michaelcalve, 2016-12-02:
Hi,

Could you give some more detail on how to setup the SD card so that when I plug it into the RPi3 it will read the files on it?

I'm not sure what's going on with "e.g.: dd if=nexmon_rpi_12-10-2016.dd of=/dev/mmcblk0 bs=1M note: the sd card device (in this example mmcblk0) may change depending on your setup"

michaelcalve, 2016-12-05:
Is it possible to use something like WireShark to capture the results and manipulate the data that's being collected from being in monitor mode!?!

DanielAW, 2016-12-05:
Hi,
I'm glad it worked for you in the end. It is possible to capture data with `tcpdump` on the RPI3 and then copy the resulting `.pcap` file to your PC where you can open it with Wireshark. Save the captured data to a pcap file using tcpdump as follows: `tcpdump -i wlan0 -s0 -w /root/mydump.pcap`.

I will close this issue for now because the original problem seems solved.

closed 2016-12-05 by DanielAW