seemoo-lab/bcm-rpi3 #27: Can't ssh into raspberry

kushagra12, 2016-12-05:
Hi,
I wanted to try the dd image but because I don't have an monitor, I could only access my RPi via ssh. When I tried ssh, it gave me 'connection refused' error. Is there any way to turn on the ssh?

DanielAW, 2016-12-05:
Hi,
our image is configured the same way the default Raspbian image is. I personally use the JTAG pins, see http://elinux.org/RPi_Serial_Connection

kushagra12, 2016-12-05:
Thanks for reply but the serial port to USB will be difficult to get where I live. Raspbian allowed ssh till November release after which we have to place a blank file named ssh in the boot directory to enable it. If you can update your dd image with the November release, I am sure I can enable ssh via same means.

DanielAW, 2017-01-09:
Closed for now due to migration to https://github.com/seemoo-lab/nexmon

closed 2017-01-09 by DanielAW