seemoo-lab/bcm-rpi3 #30: Channel Hopping for creating a file with multiple channels worth of network traffic on it

michaelcalve, 2016-12-06:
Hi so, the question that I just asked and just closed, ~an hour ago, about setting a frequency or channel # wasn't too difficult to figure out.  

I'm curious if there is a way to setup the raspberry pi and wlan0 to channel hop.  Have you or anyone ever done this before?  I'm not sure if this is a valid question for the NexMon Github Page, but I thought you all might have some more insight on this than someone new to networking, like myself!

Appreciate it!
Mike

DanielAW, 2016-12-07:
>  I'm curious if there is a way to setup the raspberry pi and wlan0 to channel hop

It basically depends on what you want to achive. For example airodump-ng already does channel hopping by default if you do not give him a specific channel as paramter. If you want to dump the wireless traffic into PCAP files you should be able to switch channels even if `tcpdump` is already running.
We implemented the `set_monitor_channel` function for the cfg80211 interface here [1]. This should allow you to switch channels as often as you like. The rest depends on the tools you use.

[1] https://github.com/seemoo-lab/bcm-rpi3/blob/master/firmware_patching/nexmon/brcmfmac/cfg80211.c#L614

michaelcalve, 2016-12-07:
> patching/nexmon/brcmfmac/cfg80211.c#L614

I've been using airodump-ng and it keeps a buffer overflow detected error
message after about 4 minutes of running.  If i write a .cap file and send
it to wireshark, prior to a buffer overload, then the file gets an error in
wireshark as it has a file with over the maximum 262144 bytes in it.

As for tcpdump, I'm unsure as to what you mean by being able to switch
channels as it is running - is there some sort of interaction command for
this??

Sincerely,
Michael Calve

michaelcalve, 2016-12-07:
And are you saying by editing the set_monitor_channel function for the
cfg80211 interface, it would be possible to set the interface to switch
channels automatically.

OR, are you saying that by calling this function in some s bash or c
script, multiple times, channels can be changed intentionally.

Sincerely,
Michael Calve

DanielAW, 2016-12-07:
You can just use `iwconfig` to set your channel

michaelcalve, 2016-12-15:
How come the 802.11n conecction speeds are not featured in the cfg802.11.c
file?

Can I add RATETAB_ENT(BRCM_RATE_150M, 0)?

Sincerely,
Michael

DanielAW, 2017-01-09:
Closed for now due to migration to https://github.com/seemoo-lab/nexmon. If you have any new infos please open a new ticket in the new repository.

closed 2017-01-09 by DanielAW