seemoo-lab/bcm-rpi3 #32: In Monitor Mode, no 802.11n packets

michaelcalve, 2016-12-15:
Hi!

Why am I not seeing any 802.11n packets in monitor mode, using tcpdump?

Is there a way to fix this?  These are the main category of packets I am looking for.

I also have a USB dongle which perhaps could be used in order to gain access to 802.11n alongside the Broadcom chip.

matthiasseemoo, 2016-12-15:
You might receive 802.11n frames, but they will not be labled as such in the radiotap header by our current firmware patch (see: https://github.com/seemoo-lab/bcm-rpi3/blob/master/firmware_patching/nexmon/patch.c#L348). If you want MCS information in your radiotap header, you can add it according to the code for the Nexus 5 (https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/monitormode.c#L65). You should also consider switching to your new nexmon repository which has a clearer code structure and supports different chips.

michaelcalve, 2016-12-15:
I am speaking more to the fact that frames being sent in the 802.11n are not being picked up, after comparing to a scan done on a macbook yesterday, which had 802.11n frames, Mbps 117.

The highest Mbps I have seen on the Raspberry Pi, also using a USB wifi network adapter, is 24 Gbps.

Also, in the cfg80211.c file for Raspberry Pi, I only see up to 54 Mbps in the static struct ieee80211_rate __wl_rates[]

michaelcalve, 2016-12-19:
> patching/nexmon/patch.c#L348). If you want MCS information in your
> patches/bcm4339/6_37_34_43/nexmon/src/monitormode.c#L65). You should also

Do you support and 2x2 MIMO chips?

We're looking to be able to capture frames at rates of 144.4 Mbps !

matthiasseemoo, 2016-12-19:
The BCM4358 of the Nexus 6P should support MIMO transmissions, but the
information is not yet in the Radiotap header that you can extend with the
information stored in the sts variable.

michaelcalve, 2016-12-19:
> issuecomment-267342093

Oh Okay, well we have been finding out that capturing all relevant, nearby
packets requires at least 2x2 MIMO for higher data rates!

Thanks for the info!

Perhaps we'll get some phone support for this idea we have.

DanielAW, 2017-01-09:
No new info, closed for now

closed 2017-01-09 by DanielAW